andreika-git/hellen-one #1: DFU does not work on a brand new board

rusefillc, 2021-02-16:
See https://rusefi.com/forum/viewtopic.php?p=39808#p39808 for a theory

rusefillc, 2021-02-16:
@andreika-git is there a resistor we can remove from hellen72b to try the fix? would it be a useful test?

andreika-git, 2021-02-16:
The only fix for Hellen72b I can think of is to carefully cut a trace between A22 and PB2 here:
[image]

andreika-git, 2021-02-16:
The solution for the next module rev: add a track jumper to the MCU module near the R128 to connect A22 and PB2 only for H7 configs. (AUX3 (A22) input cannot have a pull-up because it causes a conflict with BOOT1 on PB2 (COMP1_INP on H7))

Also, for the next Hellen72 revision we can optimize the input connections even further:
IN_CLUTCH -> RES3 (free EXTI5 and use EXTI8)
IN_EGR -> AUX3 (move from ADC3 to ADC12)
IN_NEUTRAL -> SENS3 (use EXTI5)
IN_FUEL_LEVEL -> AUX2 (move from ADC3 to ADC12)

andreika-git, 2021-02-23:
Added a tiny jumper resistor to the MCU module (not populated by default):
[image]

closed 2021-03-04 by rusefillc