andreika-git/hellen-one #2: ULN2003 flyback diode: parasitic power from main relay

andreika-git, 2021-02-16:
Problem statement: Unlike TLE6240 and other smart low-side drivers with active clamp protection, ULN2003 has a simple internal flyback diode connected to an external +12V power line. When the ECU power is off, the voltage is still present on the main relay coil (connected to the battery) and thus the V12PROT bus is powered via the flyback diode, causing parasitic power for ECU and also triggering the main relay.
The output module needs to be corrected to solve this issue.

The solution for the next revision: use an additional TVS for clamping the ULN2003 instead of the ECU power.

The temporary solution for Hellen72b:
1) Remove R314 (0.1 Ohms 1206) - to disconnect V12P from V12PROT:
[image]

2) Remove R310 and R312 - to disconnect push-pull drivers from V12PROT:
[image]

3) Connect these two pads to +12V - to reconnect the power of push-pull drivers:
[image]

rusefillc, 2021-02-17:
things are better but still three issues, @stefansf FYI

1) https://youtu.be/OEqeKp4ilVw blink if #3 plugged after #2

2) things light up BRIGHT if #2, #4, #3 inserted & ignition turned on

3) ECU does not turn off on ignition OFF

andreika-git, 2021-02-17:
Hellen72b has an additional issue with D49, you need to remove it too:
[image]

rusefillc, 2021-02-17:
coil LEDs are ON when they should not be?
ECU does not turn off

https://youtu.be/yGyNm7tmKyg

https://youtu.be/wGY7qok-m8o

https://youtu.be/YMbyt41SUfI

rusefillc, 2021-02-23:
Also related or different issue: random shut-off situation

1) turn IGN on, turn off within next few seconds: ECU shuts down instantaniously

2) turn IGN on, turn of after more than few seconds (still no cranking): ECU takes some time to shut off

3) major cranking attempts / after engine was running: ECU does not turn off

all these tests turning ON and OFF was happening with PC disconnected. there was some hot PC plugging and unplugging in test #3

rusefillc, 2021-05-19:
Done?

closed 2021-05-19 by rusefillc