andreika-git/hellen-one #9: Test mini-USB adapter board for Hellen72

andreika-git, 2021-02-19:
We have an alternative USB routed for a small adapter board near the main ECU connector. We need to test it:
- does it fit at all?
- does it work?
- can we still use both USB connectors or they interfere with each other?

rusefillc, 2021-02-20:
[image]

rusefillc, 2021-02-20:
@andreika-git what do you mean "can we still use both USB connectors or they interfere with each other?"

two cables into same PC? two cables at the same time into different PCs?

andreika-git, 2021-02-20:
Not two cables, but just one cable into any of two connectors without desoldering them

rusefillc, 2021-02-20:
that works

closed 2021-02-20 by rusefillc