andreika-git/hellen-one #14: create MAX9924 module

rusefillc, 2021-03-04:
We need a single channel VR decoder module based on MAX9924

Let's reuse a piece of https://github.com/mck1117/proteus/ as a reference schematics

rusefillc, 2021-03-04:
@andreika-git  shall we name this module 'vr-max9924" or something else?

andreika-git, 2021-03-04:
I think we should name it "vr". 
We will have 2 options for vr sensors: the built-in one already used in Hellen81 (tested on a real car, but needs more testing) and this new module "vr", that's enough IMHO.

rusefillc, 2021-03-04:
there is a chance of 9926 later if we need dual channel VR
how do we distinguish between MAX and adc-based?

andreika-git, 2021-03-04:
1) Then why not use 9926 instead of 9924 in "vr" module and make it dual-channel by default, like it was in many previous boards (Frankenso, Prometheus)?
2) Or we can use two 9924 modules for 2 channels
3) adc-based means no "vr" modules, and max-based means "vr" module - it's easy to distinguish

andreika-git, 2021-03-04:
Frankenso schematic:
[image]

rusefillc, 2021-03-04:
Let's go with 9926

rusefillc, 2021-03-08:
@andreika-git what is the status of this issue?

rusefillc, 2021-03-09:
https://github.com/andreika-git/hellen-one/commit/1248a41075824e934b88539bab09130ec319cad4 could be related

andreika-git, 2021-03-09:
The vr module was created (https://github.com/andreika-git/hellen-one/commit/1248a41075824e934b88539bab09130ec319cad4) but not used in any of the boards yet, and neither verified nor tested.

closed 2021-03-15 by rusefillc