andreika-git/hellen-one #19: chip shortage

rusefillc, 2021-03-22:
We have three tracks to address chip shortage risk

0) panic buy at least some chips? 

1) develop a F469 version of module: support F469 pinpout hoping that STM32F469IIT6 is more available due to smaller demand

2) develop 144 version of module: STM32F429ZGT6 is a bit more available _so far_

3) GD does not have 176 produces, GD 144 availability is extremely low :(

144 availability
[image]

[image]

GD

[image]

[image]

rusefillc, 2021-03-22:
[image]

[image]

rusefillc, 2021-03-23:
H743 could be the winner

https://octopart.com/search?q=STM32H743IIT6

[image]

rusefillc, 2021-03-25:
Minor update:

STM32F429IIT6 down 20%
STM32F469IIT6 down 0%, i.e. no one uses those? :)

[image]

STM32H743IIT6 availability way down

[image]

rusefillc, 2021-03-25:
https://jlcpcb.com/parts/componentSearch?searchTxt=stm32%20144

[image]

rusefillc, 2021-03-25:
Dave says https://www.win-source.net/hot-sales-and-eol-components-stm32f429zgt6.html

[image]

rusefillc, 2021-03-25:
Mouser has 144 pin chip

STM32F429ZGT6TR
https://www.mouser.com/ProductDetail/STMicroelectronics/STM32F429ZGT6TR?qs=5aG0NVq1C4wyZFvgyK4GFw%3D%3D

[image]

rusefillc, 2021-03-27:
JLC 176 change: less F4 a bit H7

https://jlcpcb.com/parts/componentSearch?searchTxt=stm32%20176

[image]

rusefillc, 2021-03-28:
[image]

rusefillc, 2021-04-10:
[image]

rusefillc, 2021-04-10:
[image]

rusefillc, 2021-04-13:
great news
[image]

rusefillc, 2021-04-14:
[image]

rusefillc, 2021-04-14:
I forgot about another option @andreika-git 

https://jlcpcb.com/parts/componentSearch?searchTxt=stm32%20208

[image]

rusefillc, 2021-04-16:
yday @andreika-git has photoshopped H7 availability but JLC tells us otherwise today

[image]

[image]

rusefillc, 2021-04-16:
[image]

rusefillc, 2021-04-19:
[image]

rusefillc, 2021-04-19:
[image]

rusefillc, 2021-04-19:
Well, even 208 goes down pretty quickly :(

[image]

[image]

rusefillc, 2021-04-21:
[image]

[image]

rusefillc, 2021-04-22:
[image]

[image]

rusefillc, 2021-04-23:
And price tripling has started :)

[image]

[image]

rusefillc, 2021-04-27:
[image]

[image]

[image]

rusefillc, 2021-04-29:
[image]

[image]

rusefillc, 2021-05-05:
[image]

[image]

still no 208

rusefillc, 2021-05-08:
Minor 176 availability increase

[image]

@andreika-git we have 208 back in stock!

[image]

rusefillc, 2021-05-20:
A lot of 144, a bit of 208 and

[image]

rusefillc, 2021-06-15:
oh boy :(

[image]

rusefillc, 2021-06-15:
144 same BAD

[image]

rusefillc, 2021-06-15:
208 is the winner today

[image]

rusefillc, 2021-06-17:
I've ordered x5 Nissan and 176 is down to qty 44. We have to start monitoring daily @BeerMoneyMotorsports

rusefillc, 2021-06-18:
:(

[image]

rusefillc, 2021-06-18:
[image]

rusefillc, 2021-06-19:
#70 created

rusefillc, 2021-06-24:
[image]

[image]

rusefillc, 2021-06-24:
[image]

rusefillc, 2021-06-24:
[image]

rusefillc, 2021-08-07:
back in stock!

[image]

rusefillc, 2021-08-24:
Got some 176 chips for $56.25 per piece :(

rusefillc, 2021-11-06:
[image]

[image]

rusefillc, 2021-11-07:
puff reminds about

https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=GD32

rusefillc, 2021-12-28:
[image]

rusefillc, 2022-02-01:
mixed news

[image]

[image]

[image]

rusefillc, 2022-03-31:
[image]

but

[image]

rusefillc, 2022-03-31:
144 is doing pretty well

[image]

rusefillc, 2022-04-06:
-100 is better today

[image]

rusefillc, 2022-05-15:
a few LOLs in the 176 department

[image]

BeerMoneyMotorsports, 2022-07-07:
i wish all were this cheap

[image]