andreika-git/hellen-one #24: Minor Bike Shed Improvement: command line gerber export?

rusefillc, 2021-04-10:
How much effort is it to export gerbers from KiCAD 5.99+ via command line?

chuckwagoncomputing, 2021-04-10:
https://electronics.stackexchange.com/questions/390135/command-line-interface-for-kicad

rusefillc, 2021-04-16:
I believe that the 7 step manual https://github.com/andreika-git/hellen-one/wiki/kicad-export-frame tells us that we really should automate the process

I've contributed first 20 bytes need help moving this forward

https://github.com/rusefi/hellen121vag/blob/main/boards/hellen121vag-a/kicad6/export_gerbers.py
https://github.com/rusefi/hellen121vag/blob/main/boards/hellen121vag-a/kicad6/step_0_export_gerbers.sh

chuckwagoncomputing, 2021-04-23:
https://github.com/JoanTheSpark/kicad-tools-1/blob/master/plot-pcb.py

Timbelom, 2021-06-02:
Two pieces of valuable info at
https://forum.kicad.info/t/command-line-to-plot-pdf/4779/11

1) there are demo scripts in kicad install folder
2) there is an example of a script which plots multiple layers while iterating over an array of parameters

Timbelom, 2021-06-02:
more examples of gerbers export https://gist.github.com/spuder/4a76e42f058ef7b467d9

Timbelom, 2021-06-02:
https://kicad-python.readthedocs.io/en/latest/ hehe

rusefillc, 2021-06-12:
@mck1117 @dron0gus can you help please? that's an important one for Hellen

chuckwagoncomputing, 2021-06-22:
Is anyone working on this? If not, I'll dig in.

rusefillc, 2021-06-22:
@chuckwagoncomputing please please please dig into! No one is working on this.

chuckwagoncomputing, 2021-06-23:
OK so the Gerbers, drill files, and VRML look like they'll be easy. I think the BOM should be easy too. I'm not sure yet about the PDF. One script I found used xdotool to automate it, as there isn't a python library to do it.

chuckwagoncomputing, 2021-06-23:
https://docs.kicad.org/doxygen-python/
https://github.com/obra/kicad-automation-scripts
https://github.com/INTI-CMNB/KiBot

rusefillc, 2021-06-23:
lets start by having 80% of the value with 20% of the effort?

chuckwagoncomputing, 2021-07-24:
https://github.com/INTI-CMNB/KiBot/issues/78

rusefillc, 2021-08-19:
@Come1LLF00 can you please start downloading https://kicad-downloads.s3.cern.ch/windows/nightly/kicad-msvc.r23615.dfacac5eb6-x86_64.exe from https://kicad-downloads.s3.cern.ch/index.html?prefix=windows/nightly/

rusefillc, 2021-08-19:
C:\Program Files\KiCad\5.99\share\kicad\demos\python_scripts_examples\gen_gerber_and_drill_files_board.py

chuckwagoncomputing, 2023-03-15:
I've made some progress:
https://github.com/chuckwagoncomputing/hellen121vag/tree/export-script
I added some scripts that I suppose should end up in this repo once they're fully working.
https://github.com/chuckwagoncomputing/hellen121vag/blob/export-script/export.sh
https://github.com/chuckwagoncomputing/hellen121vag/blob/export-script/fill-zones.py
https://github.com/chuckwagoncomputing/hellen121vag/blob/export-script/export-vrml.py
They produce all the correct files now, but I'm not sure yet if they produce them correctly.
This requires Kicad 7.0.1. I'm using kicad-cli, which was introduced in 6.99/7, and one of the necessary flags was broken until 7.0.1.

chuckwagoncomputing, 2023-03-17:
Current status is: wait for https://gitlab.com/kicad/code/kicad/-/merge_requests/1562 to be accepted and released.

rusefillc, 2023-03-17:
Wow.

chuckwagoncomputing, 2023-03-17:
Fun fact, in Gitlab, CI jobs for merge requests run on whatever runners the requester has set up.
Kicad self-hosts runners that build in about 10 minutes. (probably because they don't do a full rebuild)
The default Gitlab runners seem to take somewhere close to three hours. When creating a merge request, one of the stated prerequisites is to increase the build time limit in the settings of your for to >3h, from the default 1h.

chuckwagoncomputing, 2023-03-22:
Kicad change has been merged, should come out in 7.0.2 which is scheduled to be released April 13.

rusefillc, 2023-03-22:
@chuckwagoncomputing I would say proper location for those is not in the individual board repo but in the toolset hellen-one repo, something like hellen-one/kicad/bin?

chuckwagoncomputing, 2023-03-22:
Yes, there are nightly KiCad builds. However, I think we should wait to transition to KiCad 7 until 7.0.2 is available as a stable release.

I've put the scripts in hellen-one/bin
https://github.com/chuckwagoncomputing/hellen-one/tree/export-scripts/bin

Should I PR now? It shouldn't hurt anything having the scripts present.

rusefillc, 2023-03-22:
Yes happy to merge as is

rusefillc, 2023-06-01:
@chuckwagoncomputing looks like KiCad Stable Release Current Version: 7.0.5

Still very interested in all this!

chuckwagoncomputing, 2023-06-02:
Why does the create-board.yaml workflow use a Docker image, and should the new scripts be run in the same Docker container?

rusefillc, 2023-06-02:
@chuckwagoncomputing docker was a wah to address fragile setup - without it we were in dependencies management hell.

I would like to have scripts run both ways: no docker so that we can test and troubleshoot and docker for... actually not sure if we need docker for export phase maybe we fire docker only for Gerber merge later

chuckwagoncomputing, 2023-06-03:
Well apparently my change is somehow not in the release version of Kicad. I don't know what's up with that.

rusefillc, 2023-06-08:
Could be up a month or more until next 7 release :( see https://www.kicad.org/blog/

@chuckwagoncomputing has been added as a collaborator with write permissions on the https://github.com/rusefi/build_support/ repository.  

@chuckwagoncomputing shall we push any nightly into usefi/build_support/ or else if that would help to move forward?

rusefillc, 2023-07-13:
https://github.com/andreika-git/hellen-one/issues/307 created to discuss that specific detail

rusefillc, 2023-07-13:
https://github.com/andreika-git/hellen-one/issues/308 created for a usability improvement

rusefillc, 2023-07-15:
FYI, low priority: trying to use the script just to export gerbers and got seg fault invalid input parameters https://github.com/rusefi/rusefi-hardware/actions/runs/5559501944/jobs/10155684390

rusefillc, 2023-07-23:
I forgot about BOM https://github.com/andreika-git/hellen-one/issues/313

rusefillc, 2023-07-23:
> BOM

https://github.com/andreika-git/hellen-one/issues/313

rusefillc, 2023-08-02:
@chuckwagoncomputing FYI usability improvement ask #317 - not really important but might be an easy fix

rusefillc, 2023-08-02:
@chuckwagoncomputing one more #320

rusefillc, 2026-01-04:
this works great now!

closed 2026-01-04 by rusefillc