andreika-git/hellen-one #26: h64 CPL rotation

rusefillc, 2021-04-10:
[image]

[image]

andreika-git, 2021-04-11:
https://github.com/andreika-git/hellen-one/commit/6ed466df55011effc9942c4281bbaf00b0e35188
https://github.com/andreika-git/hellen-one/commit/46ce17a2ef54e39abf2cab8c6a52183a04fd5818

It's better now but still needs more thorough check:
[image]

When reloaded, it shows overlay with the board image, but with some vertical offset:
[image]

The whole component layer is shifted:
[image]

I don't know whether it's the JLC's bug or our bug... 

Also the USB connector is shifted horizontally (upper corner) - possibly wrong CPL origin set in the Kicad footprint.

closed 2021-04-11 by andreika-git

andreika-git, 2021-04-11:
Status update:
- gerber viewer coordinate test shows there's no noticeable vertical offset. Assuming that's JLC's bug
- USB connector footprint (Hellen-one-common Kicad lib) has wrong origin: it uses 1st pad center instead of component centroid.