andreika-git/hellen-one #30: h72 LEDs as reference resistor values

rusefillc, 2021-04-20:
all red are perfect
all blue are perfect
all white are too bright need to get much dimmer
12v green is OK-ish, maybe just a bit dimmer please?
3 and 5v greens are too bright, needs to get much dimmer

rusefillc, 2021-04-20:
https://youtu.be/RHhSn1nypz4

Qwerty-OFF, 2021-05-01:
Calculated values

GREEN:
12 V - 5.1 kOhm
5 V - 1.6 kOhm
3.3 V - 750 Ohm

WHITE:
12 V - 19 kOhm
5 V - 5 kOhm
3.3 V - 1.6 kOhm

rusefillc, 2021-06-16:
@andreika-git please move to wiki

andreika-git, 2021-06-16:
Please elaborate - these R values are for 0603 LEDS or 0805 LEDs? They have different Vdrop/intensity.

rusefillc, 2021-06-16:
@Qwerty-OFF please move content to https://github.com/andreika-git/hellen-one/wiki

rusefillc, 2021-10-16:
See also https://github.com/rusefi/hellen154hyundai/issues/33

rusefillc, 2021-10-16:
See also https://github.com/rusefi/hellen154hyundai/issues/34

rusefillc, 2022-04-24:
@Qwerty-OFF status update: nissan rev C

WBO module red is perfect reference point

top side of board red is brighter than WBO module red, could be reduced to WBO level of red

green ones at test point are EXTREMELY bright, needs major reduction

[image]

Qwerty-OFF, 2022-08-22:
|LED|3v3|5v|12v|Status|
|---|---|---|---|---|
|GREEN 0805 (C2297)|5.1k|20k|100k|Need to check on Hyundai 154|
|RED 0603 (C2286)|1k|   |   |   |
|TBD|   |   |   |   |