andreika-git/hellen-one #31: h72 schematics usability

rusefillc, 2021-04-21:
[image]

That's very confusing.

Do we have two different "in_knock" here - module internal "in_knock" pad and h72 frame "in_knock" net? 

If those are in fact two different entities having same name I would suggest to change the naming convention to avoid the confusion. I like the "raw" for inputs from connector, maybe "mcu_knock" for all nets between MCU and modules?

andreika-git, 2021-04-21:
IMHO this should belong to https://github.com/rusefi/hellen72/issues/

rusefillc, 2021-04-21:
I bet that's a hellen-one pattern not an issue specific to any board

andreika-git, 2021-04-23:
[image]

rusefillc, 2021-04-23:
👍

closed 2021-06-16 by rusefillc