andreika-git/hellen-one #32: WBO module needs SWD exposed as side pads on the module

rusefillc, 2021-04-24:
very few people have TC2030 plug

to increase usability we would need SWD as a 2.54mm header 

we do not want to add  2.54mm header on the WBO module due to space constraints, we want SWD exposed as side pads on the module so that specific board could do whatever

rusefillc, 2021-04-24:
not sure how standard is this https://wiki.paparazziuav.org/wiki/STLink

rusefillc, 2021-04-24:
https://github.com/rusefi/hw_microRusEfi/raw/master/compatible_ST-LINK_V2.jpg is the fake v2 we use on microRusEFI

mck1117, 2021-04-24:
Do we need availability? 99% of people with MRE/Proteus don't have any programmer. We only need programming interface for "factory programming"

rusefillc, 2021-04-24:
A number of boards are open source, also while 042 is poorly available there is too much of a chance of units shipped without 042

rusefillc, 2021-05-19:
Done

closed 2021-05-19 by rusefillc