andreika-git/hellen-one #39: NB2 tach HW magic. Maybe related to alternator.

rusefillc, 2021-05-04:
Someone please write this up?

rusefillc, 2021-05-04:
CrazyStriker says remove Q313, R313 and D304

[image]

rusefillc, 2021-05-05:
[image]

rusefillc, 2021-05-05:
Q313 removal has returned tachometer into my life

https://youtu.be/WFEwe_KtHW4

Qwerty-OFF, 2021-05-05:
Maybe fix protection like it made in power12 module?
[image]

mck1117, 2021-05-05:
Either fix it like that, or just remove it entirely.  I don't understand the reasoning to keep it around if the one in power12 exists too.

closed 2021-06-16 by rusefillc