andreika-git/hellen-one #42: don't use LEDs as a pullup

mck1117, 2021-05-06:
If you want an LED and a pullup, wire both.  Do not rely on the LED as a pullup.  Using the LED in series with the pullup may not have the desired effect.

For example, in the VR module:

[image]

andreika-git, 2021-05-06:
We do not rely on LED resistors for pull-ups. We do have pull-ups in both input and dinput4 modules (for all required inputs including VR):
[image]

closed 2021-05-06 by andreika-git

mck1117, 2021-05-06:
Then what's going on in the VR module?

[image]

andreika-git, 2021-05-06:
What exactly is wrong in the VR module? The "true" pull-up resistors are not in the VR module, they are in the input module because the VR module is not the only source of VR signal - Hellen One has a working software VR processing capability.

mck1117, 2021-05-06:
> The "true" pull-up resistors are not in the VR module

this is antithetical to the whole point of a modular design

> VR module is not the only source of VR signal

If you are suggesting connecting more than one source to the same net (then connecting to a single MCU pin), why? We can switch inputs in software.  One source.  One sink.  One pullup, located on the schematic of the SOURCE.

> Hellen One has a working software VR processing capability

I'll believe it when I see it

mck1117, 2021-05-13:
this has now caused confusion in #49 - this module really does need internal pullup so that use of an extra module isn't needed.