andreika-git/hellen-one #43: revise power5v input capacitors

mck1117, 2021-05-06:
At 50-60% of the rated voltage (12-15v on a 25v rated cap), a typical x5r/x7r ceramic loses 50-70% of the rated capacitance.  That means the spec'd 40uF ends up closer to 10-15uF.

One of the following needs to happen:
- Replace ceramics with a single aluminum polymer electrolytic (bonus: the ESR will probably reduce conducted emissions)
- Add significantly more capacitors to have enough capacitance when at 12-15v.

As an aside, 25v rated caps really aren't appropriate here.  A load dump could result in more than that (even with the clamping circuits in place), and the failure mode of ceramics is generally to fail short, which will cause secondary failures.

andreika-git, 2021-05-06:
We do not use 25v rated caps, we do use 50v rated caps:  CL31A106KBHNNNE 10uF 50V 1206 (C13585).
[image]
[image]
[image]

closed 2021-05-06 by andreika-git

mck1117, 2021-05-06:
Ah, ok, I was looking at the wrong BOM line.  Even then - it still loses ~25% of the capacity at 12-15v, so it's still worth adding one or two more caps.