andreika-git/hellen-one #44: [output] clamp V12FLYBACK to supply rail

mck1117, 2021-05-06:
Clamping energy should be dumped back to the supply rail.   Dumping it in to a TVS is asking for the TVS to overheat and fail, which then leaves them unclamped, which will cause the driver chip to fail.

andreika-git, 2021-05-06:
How exactly we should dump the energy back to the supply rail? We already had it connected to the supply rail for Hellen72 rev.A - it was an obvious but *wrong* solution.
[image]

mck1117, 2021-05-06:
Why was it a wrong solution? Dump it back to the main relay switched pin.

andreika-git, 2021-05-06:
It's wrong because we have a situation when the voltage is present on output pins through the relay coils ("hot at all times"), but not on the board power rail (the ignition is turned off). So we had a parasitic power through the flyback diode.

mck1117, 2021-05-06:
why is there a connection between the main relay supply pin (switched by the main relay) and input to the regulator?

mck1117, 2021-05-06:
Actually, why not dump back in to the 12V_PERM net? That's not switched anyway, so it's guaranteed to be unable to keep the ECU on.

andreika-git, 2021-05-06:
Because one of the use-case configurations is to power the ecu board from the main relay supply (eg. finish some work even after the ignition switch is off). 
Yes, we can add a diode to 12V_PERM, but it's not a universal solution (and thus unsafe): 12V_PERM is not guaranteed to be present on all car pinouts.

closed 2021-05-06 by andreika-git

mck1117, 2021-05-06:
> but it's not a universal solution (and thus unsafe)

therefore the solution shouldn't be inside the module - the module should expose as its API a pin for "this pin may need the car's implementation to sink current", which could be a diode to some supply pin, or could be a TVS.