andreika-git/hellen-one #45: [sd] add decoupling cap for SD card

mck1117, 2021-05-06:
SD card should have a decoupling cap across the power rails near the socket.  I've heard stories (not our boards) of people having SD card unreliability due to no decoupling caps near the card.

andreika-git, 2021-05-06:
We already have more than enough decoupling caps for the sd card on the sd module:
[image]
[image]

closed 2021-05-06 by andreika-git

mck1117, 2021-05-06:
Apparently my eyes can't read the monochrome schematics from Altium.

22u is pretty big.  Absolutely worth replacing one of the 22u with a 1u or even 100n.