andreika-git/hellen-one #47: h72 fails short circuit at 22v supply

mck1117, 2021-05-06:
Some background:
- I once saw 19 volts on a real car with a failing alternator.
- Three additional volts is not out of the question.

I connected h72b to a bench supply (with current limiting set to ~1A), and slowly increased the supply voltage.  At around 22 volts, I heard a pop, and the supply went in to limiting.  Turning the voltage down, the resistance of the fault was on the order of 0.5-0.7 ohm.  Removed the TVS, TVS measured fine.  Removed one of the crowbar mosfets, worked fine.  Removed the other, short circuit!

Last time we visited load dump protection, you said:

> This is a basic built-in solution which protects only from “Test B" (ISO 16750-2), i.e. with centralised load dump suppression.

But test B specifies 35 volts for 10-100ms.  So does Hellen expect to survive 16750-2 test B or not?

ps: did that issue get deleted? I still have it in email 🙂

rusefillc, 2021-05-06:
h72c uses a PS much closer to proteus

mck1117, 2021-05-06:
> h72c uses a PS much closer to proteus

yes - the power5 module should be ok, the problematic crowbar is in the power12 module, which is essentially the same between b and c

rusefillc, 2021-05-06:
> ps: did that issue get deleted? I still have it in email 🙂

old issue was removed while this hellen-one repository was re-created. @mck1117 please copy-paste anything useful from https://github.com/andreika-git/hellen-one-old/issues/3

Qwerty-OFF, 2021-05-06:
It seems the Hellen 72 does not contain a fuse. Since a limited current PSU was used, the 5 amp fuse would not blow even if it was installed. The transistor will work as a zener diode dissipating power P = (Vin-20V) * I, i.e. about 2W in the above situation. This is a lot for the sot-23 case.

We have the following solutions:
- Do nothing, assuming the faulty alternator is supplying enough current to blow the fuse (and add the fuse to h72).
- Replace the protective transistor with a more powerful and heat-intensive one, so that the fuse burns out earlier than the transistor.
- Implement non-"crowbar" overvoltage protection.

If we want to pass the test described above, we must select one of the first two options. If we really want to survive a faulty generator, the last option is most likely worth choosing.

mck1117, 2021-05-06:
> The transistor will work as a zener diode dissipating power P = (Vin-20V) * I, i.e. about 2W in the above situation. 

It's even worse than this.  At 22v the drop thru the tvs is ~19 volts plus the plain reverse protection diode, leaving ~2 volts on the gate of the transistor - according to the datasheet that's enough to support 15-20 amps of drain current, more like 350 watts peak.  A SOT23 will tolerate 2W for a pretty long time - the AO3400 is even rated for that dissipation for multiple seconds.  But at 350 watts, the limit is something like 10 microseconds (though 350 watts is also awfully close to the "never exceed for any period of time" dissipation).

[image]

> Replace the protective transistor with a more powerful and heat-intensive one, so that the fuse burns out earlier than the transistor.

yes, this is what I was getting at - paralleling mosfets doesn't really work (variation in gate voltage means a single transistor takes the full hit), so the mosfet will weld itself in to a short long before a 3/5/10 amp fuse pops.  It will then blow the fuse, but is still shorted.  Replace the fuse, ECU is still shorted.

As for what solution to implement:  The power12 module already has a TVS rated for 1500 watts peak, which is PLENTY of power to pop probably even a 20 or 30 amp fuse.  With the revised power5 module tolerant of up to 40v input, it's fine to just remove the crowbar entirely.

[image]

It is designed to tolerate massive single-event pulses, like the one required to blow a fuse.  At the same 350 watts, the maximum pulse is tens of milliseconds - plenty of time to blow a fuse.

Qwerty-OFF, 2021-05-07:
Output module contains RZ7899 (25V max), TC4427 (22V max) and high-side:
[image]
[image]
Did I understand correctly that this high-side should not survive more than 12V?

closed 2021-05-24 by andreika-git