andreika-git/hellen-one #49: PWM vr-interface module

rusefillc, 2021-05-06:
Let's try https://github.com/mck1117/vr-interface as an untested-on-real-vehicle alternative to MAX9926 VR conditioner

Is that vr-interface large enough for a module or do we copy/paste it as is?

rusefillc, 2021-05-06:
@mck1117 would you be interested to practice your Hellen module skills?

mck1117, 2021-05-06:
I think it warrants a module, yes.  Go ahead and assign to me.

rusefillc, 2021-05-06:
I am hoping that we are 1 week from sending hellen-mercedes to the fab. I guess technically we can totally skip PWM vr-interface in rev A but see what you can do please?

mck1117, 2021-05-11:
yes, that looks OK to me.  The unwritten (in the schematic) part of the NOR gate that we need is that it must allow you to sink current out of the input pins - using the diode from GND -> input.  However, the replacement chip allows this in the datasheet, so everything looks OK.

rusefillc, 2021-05-13:
@mck1117 which of the three outputs do we want to feed into Hellen?

[image]

mck1117, 2021-05-13:
"Output" = digital output, conditioned VR signal, actual trigger input.
"Analog out" = if we want an oscilloscope trace of the trigger
"Thresh in" = pwm output from stm32

rusefillc, 2021-05-13:
sounds like "Analog out" is good to have a via on the PCB for just in case but not route to MCU

mck1117, 2021-05-13:
I want to route it to ADC3 so we can do this: http://www.maxxecu.com/webhelp/settings-diagnostics-trigger_oscilloscope.html

rusefillc, 2021-05-13:
The knock ADC3? :)

mck1117, 2021-05-13:
Yes, that ADC3 🙂

We don't ever need knock and trigger scope at the same time.  Trigger scope with some free revving to check the trigger threshold, then knock once you're actually driving around.

rusefillc, 2021-05-13:
@Qwerty-OFF @andreika-git  ^^^

Qwerty-OFF, 2021-05-13:
[image]
Is it ok?

mck1117, 2021-05-13:
@andreika-git does `vr-0.1` need external pullup resistor? This is exactly the confusion in #42.

mck1117, 2021-05-17:
@andreika-git ping

andreika-git, 2021-05-17:
> 
> Is it ok?

Yes, it's ok, no additional pullups needed.

mck1117, 2021-05-17:
No pullup or not? Because in #42 you said the VR module requires an external pullup.

andreika-git, 2021-05-17:
I didn't say anything about "external" pullup. I said that the required pullup for crank-vr input is already in the respective module. No additional pullups required for the schematics done by @Qwerty-OFF.

rusefillc, 2021-05-19:
Created!

closed 2021-05-19 by rusefillc