andreika-git/hellen-one #63: opamps should prefer MCP6004

mck1117, 2021-05-28:
TSSOP14 opamps should prefer MCP6004T-I/ST (C50282) over LMV324, as the DC performance of the MCP is better.  In particular, the LMV can't get within ~0.5v of the positive rail, which is required for some OEM TPS sensors (et al)

rusefillc, 2021-06-21:
See #27

rusefillc, 2021-07-08:
We need new revisions of both Hellen MCU 144 and 176 with C398929 and LMV324 completely wiped out TODAY

Very disappointing that we have not handles this better and now we have at least 10 boards full of LMV324, this was very preventable.

andreika-git, 2021-07-08:
https://github.com/andreika-git/hellen-one/commit/6e68005fa9b13a718671ad632f19f5b2b39c0ed1

closed 2021-07-08 by rusefillc

andreika-git, 2021-07-29:
Another advantage is that MCP6004T can withstand up to -1V negative input, but LMV324 is limited to only -0.2V, which is less than SRV05-4 limiter can guarantee. LMV324 saturates output for even slightly negative input voltages.

mck1117, 2021-07-29:
lmv even saturates the output with 0v input