andreika-git/hellen-one #67: Injector Driver - yet another option is out of stock

rusefillc, 2021-06-16:
ZXMS6005SGTA_C174042 is out of stock

is BSP76E6433_C152392 an option or too slow?

@mck1117 @andreika-git

mck1117, 2021-06-16:
ouch

I /think/ that should be ok, if a little slow

rusefillc, 2021-06-17:
@blundar what's your take on https://datasheet.lcsc.com/lcsc/1810172210_Infineon-Technologies-BSP76E6433_C152392.pdf as injector driver?

rusefillc, 2021-06-20:
@mck1117 says he is aware of more legit options

rusefillc, 2021-08-19:
1000 purchased for NRS

closed 2021-08-19 by rusefillc

rusefillc, 2021-11-07:
#111

rusefillc, 2022-01-01:
is IRF7301TRPBF a good chip for us? we have not used this pinout so far but on the bright side very available

https://octopart.com/search?q=IRF7301TRPBF&currency=USD&specs=0

blundar, 2022-01-01:
That looks like a standard, unprotected MOSFET.  It's not a protected
driver and shouldn't be used as such.  Depending on use case, may or may
not be fine.

blundar, 2022-01-01:
If you're not going to be using a protected driver, I'd be looking at more
like 50-100Vds rated parts and adding an active clamp circuit to the base
of the FET with clamp voltage 2/3 to 3/4 the rated Vds.  If you need short
circuit protection, you'd also need to additionally either add a shunt
resistor or use the Rds(on) of the FET in addition to a current sense
amplifier.
-D