andreika-git/hellen-one #71: What chips should go into NRS

rusefillc, 2021-06-20:
We need National Rusefi Stockpile

Just ordered x25 of STM32F427ZIT6 since newark still had some stock

what are the next chips to go into NRS?

injector driver? what PN from mouser digiket?
what else is weird and will get harder to get?

rusefillc, 2021-06-20:
See #19

andreika-git, 2021-06-20:
**Short list:**
STM32F042K6T6 (C69216)
LMR14030SDDAR (C182078)
AP2337SA-7 (C150497)
AUIPS2041LTR (C113610)
BTS3028SDR (C112639)
L6205PD (C36543)
LPS25HBTR (C87898)
***

**Long list (sorted by JLC stock):**
LIS2DW12TR (C189624) 111
TJA1051T/3/1J (C38695) 631
PESD1CAN (C343993) 798
AD8628ARTZ (C29309) 1114
SMCJ30A (C310046) 1315
MCP6004T-I/ST (C50282) 1925
TC4427AEOA713 (C20551) 2162
REF3033AIDBZR (C36658) 3741
MIC5205-3.3YM5 (C37970) 3953
RZ7899 (C92373) 16298
PSM712-LF (C32677) 28098
XC6206P332MR (C5446) 25601

rusefillc, 2021-06-20:
I am super confused

[image]

[image]

rusefillc, 2021-06-20:
@andreika-git @qwerty-off LMR14030SDDAR is less available but is LMR14030SSQDDAQ1 is more available

what's the difference?

Qwerty-OFF, 2021-06-21:
ISL9V3040S3S (coil IGBT)
MAX9926 (VR)

rusefillc, 2021-08-10:
ordered LMR14030SDDAR  and ISL9V3040S3S

rusefillc, 2022-03-04:
@andreika-git @Qwerty-OFF I've just ordered some BM2LC105FJ-CE2

those are compact two channel injector drivers could be useful on our smaller boards like Harley or alpha etc