andreika-git/hellen-one #76: AP2331W-7 alternatives

rusefillc, 2021-06-28:
U103 is AP2337SA-7 or AP2331W-7

I have the last 62 available for sale on this small green Earth.

Are there alternatives?

rusefillc, 2021-06-28:
Merc uses two "U103, U480"
H72 NB2 uses same two

rusefillc, 2021-06-29:
@andreika-git please clarify
```
AP233* is not required, it's optional. The board runs just fine without it.
let's confirm that this applies to both part numbers since some board have two of it
```

andreika-git, 2021-06-29:
U103 (mcu) AP233* is not required, it's optional. The only reason for AP233* is to have a "close-to-5V voltage" when powered from USB-only. Otherwise we have a diode D101 which is a backup replacement in case of AP233* is not populated, but at the cost of voltage drop. The same applies to U480 (stlink) and D480 backup.
[image]
[image]

rusefillc, 2021-06-29:
@mck1117 would be interested to find an alternative part anyway just for fun?