andreika-git/hellen-one #79: inconsistent DFU

rusefillc, 2021-07-11:
We have reports of two or three kinds of Hellen boards failing to reboot into DFU via "reboot_dfu" command

Same reboot to DFU procedure could be initiated by rusEFI console using "Program" button while "Automatic DFU" is selected

This issue is NOT about "reset" or "PROD" button, the scope of this issue is ONLY software-based reboot to DFU

Steps to reproduce:
1) have a Hellen board with 407 chip
2) have Hellen board with 407 blink blue LED
3) connect via USB
4) hit "automatic DFU" in rusEFI console

expected result: DFU process starts

actual result: board reboots into normal operation

BeerMoneyMotorsports, 2021-07-11:
Auto DFU tests USB power only

H72c w/F429- Pass-5 Fail- 0
H64a w/F407- Pass-0 Fail- 5
H64b w/f407- Pass-1 Fail- 4

[image]

BeerMoneyMotorsports, 2021-07-11:
H72c
[image]

H64a
[image]

H64b
[image]

rusefillc, 2021-07-11:
[image]

not hellen with old 407 SW dfu jump works!

that's inconclusive

i also have 2nd 176 407 from same pack of samples

that opens many possibilities but i would compare designs arounds caps etc before we go swapping parts

maybe up some caps?

@dron0gus

mck1117, 2021-07-12:
With the PROG button held (ie, MCU always enters system ROM at reset), the reset line is triggering every exactly 1 second, so something is triggering a hardware reset after ~1 second 

Reset is low, not reset is high:

[image]

The large period at the left is me pressing the physical reset button, then the ~1 second spaced blips are the MCU resetting itself.

mck1117, 2021-07-12:
[image]

mck1117, 2021-07-12:
When attaching a debugger to a completely blank chip when boot looping the bootloader, it says the software reset bit is set in RCC register CSR.

[image]

mck1117, 2021-07-12:
Here are an old MRE (f407), f4 disco, and hellen probed at the OSCOUT (PH1) pin:

hellen:

[image]

microRusEfi:

[image]

F4 discovery:

[image]

rusefillc, 2021-07-12:
OMG

[image]

rusefillc, 2021-07-12:
```
I replaced $0.2 part with a "crystal and caps" that is guaranteed to oscillate well within the allotted time (edited) 
the bootloader waits 800us for the clock to start
if the clock doesn't start in 800us, it triggers a system reset
our clock is locking in ~900us or so
```

mck1117, 2021-07-12:
The above photo is injecting an 8mhz sine wave in to the OSC_IN pin - so that we can guarantee the "crystal" will have locked

mck1117, 2021-07-12:
Evidently the F42x bootloader waits longer, because on a random proteus board it takes even longer, but the bootloader still works perfectly

mck1117, 2021-07-12:
With the F407 discovery's oscillator transplanted on to the hellen121a PCB:

[image]

andreika-git, 2021-07-12:
There are some JLC drop-in alternatives to consider: 
* C122522
* C172361
* C111120
* C90881

andreika-git, 2021-07-12:
Here's a similar topic already discussed:
https://community.st.com/s/question/0D53W000004lc5NSAQ/dfu-over-usb-not-working-hse-not-detected

mck1117, 2021-07-12:
This will add support for alternate crystal speeds without different firmware: https://github.com/rusefi/rusefi/pull/2952

rusefillc, 2021-07-12:
520-200-12-30-AEM-TR

rusefillc, 2021-07-12:
Maybe this is linked to the embedded bootloader version:

V3 on STM32F407 and STM32F401
V7 on STM32F427
V9 on STM32F446

Check AN2606 for details

https://www.st.com/resource/en/application_note/cd00167594-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf

rusefillc, 2021-07-12:
digikey stuff in stock

H130A-8.000-8-2030-TR‎ 8 MHz
‎2845-WTL5G60390VH-ND‎ 8 MHz
535-9105-1-ND (20 MHz ESR 30 Ohm)
CTX1436CT-ND (16 MHz, ESR 40 Ohm)
644-1037-1-ND (16 MHz, ESR 120 Ohm)

mck1117, 2021-07-12:
Ordered a stack of 16 and 20MHz options from digikey, will test soon on 121vw

rusefillc, 2021-07-14:
@BeerMoneyMotorsports please order qty 20 of 

ABM3-20.000MHZ-B2-T  535-9105-1-ND 20 MHz ESR 30 Ohm

I believe that we have major improvement in Hellen72 rev D behavior

rusefillc, 2021-07-15:
@mck1117 says that 20MHz crystal ABM3-20.000MHZ-B2-T starts in ~250us

looks like minimal fix for Hellen MCU is to jump from 8MHz to 20MHz and move on?

@andreika-git @mck1117

closed 2021-08-19 by rusefillc