andreika-git/hellen-one #82: VAG ibom rendering issue

rusefillc, 2021-08-06:
good example https://rusefi.com/docs/ibom/hellen81-a-ibom.html rendered from https://github.com/andreika-git/hellen81/
good example

problem statement https://rusefi.com/docs/ibom/hellen121vag-b-ibom.html

just a theory: kicad export VRML - maybe in case of large connector we have broken zero point reference and we have offsets drama? 

ibom-data folder the NOT kicad folders maybe we are not handling great? maybe improvements for error handling?

andreika-git, 2021-09-13:
We have a temporary solution for the "component offset" drama:
https://github.com/andreika-git/hellen-one/commit/5dd37131639f241db01b9e4a0ce3106a9ebd4e90

rusefillc, 2021-09-14:
We are affected by KiCAD VRML export ignoring "do not show 3D model"

IMHO the proper thing to do would be to create an issue at https://gitlab.com/kicad/code/kicad/-/issues

@andreika-git can you please create an issue at https://gitlab.com/kicad/code/kicad/-/issues let's see how that goes

andreika-git, 2021-09-14:
[image]

andreika-git, 2021-09-14:
Do we need to check on more recent Kicad version?
[image]

rusefillc, 2021-09-14:
https://gitlab.com/kicad/code/kicad/-/issues/9166

andreika-git, 2021-09-14:
Here's why the h128merc board image is dark:
[image]

rusefillc, 2023-03-07:
Let's revisit with KiCAD 7?

andreika-git, 2023-03-20:
Changed board thickness to 1.6 mm hoping to resolve the issue.