andreika-git/hellen-one #101: L6205N PowerDIP not PowerSO

rusefillc, 2021-10-09:
We have exhausted L6205PD stock at JLCPCB
While I am trying to snatch remaining tube packaged L6205PD we have to consider the L6205N PowerDIP option for two reasons

1) JLCPCB might be able to acquire L6205N for us
2) if not, PowerDIP would be 400 times simpler to solder on US soil

See https://datasheet.lcsc.com/lcsc/1810241722_STMicroelectronics-L6205PD013TR_C36543.pdf

rusefillc, 2021-10-09:
@blundar @mck1117 @andreika-git opinions?

blundar, 2021-10-09:
When you have shit to deal with, remember that it is s fertilizer and plant
crops.

andreika-git, 2021-10-09:
@andreika-git suggests real-life testing on a prototype board

rusefillc, 2021-10-09:
wow mouser has a few of EVAL6205N

[image]

blundar, 2021-10-10:
Thermal performance is going to be shit compared to powerso.  They do make
heatsinks that can be used, both clip on and glue on.  Id probably
recommend a thermal adhesive for this because vibration.  The #1 case where
problems will occur is not a current RE test vehicle: centrifugal blower
(vortec and procharger) at medium (25-50%) throttle with a large throttle
plate.  In this murder case, you have the worst case high velocity airflow
and the case where you’re trying to maintain a steady throttle with strong
forces (from air velocity).  You’re going to be at significant duty for
sustained periods of time.  Duty x time = thermal load.

I’m 100% convinced you will be able to make this chip run an ETB.  You’re
going to run into trouble on the endurance race.  You’re going to run into
trouble with demanding ETB applications like my 102mm TB + D1sc SC
nightmare example.  Does this matter for what you’re going to use it for?
Maybe.  If you want help torturing your evaluation boards appropriately,
lmk.

Dave