andreika-git/hellen-one #111: BSP75N C114056  SOT-223 POWER DISTRIBUTION SWITCHES ROHS

rusefillc, 2021-11-07:
is BSP75N  an option for us as injector driver?

is it compatible with current injector driver?

[image]

there are four related lines

https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=BSP75N

[image]

rusefillc, 2021-11-07:
@andreika-git @mck1117 @qwerty-off?

rusefillc, 2021-11-07:
We've mentioned BSP76 previously but not BSP75N

rusefillc, 2021-11-07:
look they even say solenoid!

[image]

rusefillc, 2021-11-07:
low current rating? :(

[image]

mck1117, 2021-11-07:
That's too low for an injector or VVT.

rusefillc, 2022-03-17:
https://github.com/andreika-git/hellen-one/wiki/module-output#parts-options

closed 2022-03-17 by rusefillc