andreika-git/hellen-one #118: resistors around TC4427

rusefillc, 2021-11-23:
there is an opinion that more resistors are needed to save stm32

https://github.com/rusefi/rusefi/issues/3558

[image]

rusefillc, 2021-11-23:
we suspect that this board had wild alternator spikes due to bad PID settings this might have contributed to problem

rusefillc, 2021-11-25:
@mck1117 do we have similar RC on any other HW?

mck1117, 2021-11-25:
Proteus and MRE both have it on their ignition/pushpull outputs.

mck1117, 2021-11-25:
Actually I take that back - Proteus/MRE have pulldown resistors on the input side, but no series resistor.  They do, however, have a resistor on the output, also absent from Hellen.

[image]

andreika-git, 2021-11-26:
All Hellen boards also have pull-down resistors:
[image]
But no series resistors.

andreika-git, 2021-11-26:
Done: output-0.3
https://github.com/andreika-git/hellen-one/commit/520bed03c96fdb8829f21b9a2b18f8b3528e3318
[image]

closed 2021-12-02 by rusefillc