andreika-git/hellen-one #119: do we survive 5va output power to GND?

rusefillc, 2021-12-02:
See https://github.com/rusefi/rusefi/issues/3559

rusefillc, 2021-12-02:
First order of business is to document current state on MCU module wiki page?

rusefillc, 2021-12-03:
I have about 400 of TLS115D0EJXUMA1

let's please move to TLS115D0EJXUMA1 those are cheap and very resilient

rusefillc, 2021-12-13:
Documented on https://github.com/andreika-git/hellen-one/wiki/module-mcu#supply-output

closed 2021-12-13 by rusefillc

rusefillc, 2022-01-07:
at this point we do NOT have a unified hellen solution - we simply slap TLS115 on all boards individually. maybe that should be revisited but that's the way it is right now

@beermoneymotorsports FYI

rusefillc, 2022-01-07:
reopening:

@andreika-git what do you think about adding TLS115 on the back of MCU module to make a uniform fix for more of hellen boards?

andreika-git, 2022-01-07:
> reopening:
> 
> @andreika-git what do you think about adding TLS115 on the back of MCU module to make a uniform fix for more of hellen boards?

Adding TLS115 on the back is hard on its own, but also it requires some additional conditions:
[image]
Two more capacitors - hand soldering on the back side?
And +12V source which is unobtainable inside the mcu module.

Maybe we can just try replacing the L101 inductor with a resettable PTC 0805 fuse. For example, SMD0805-050_C66452, which has 0.5A hold current rating and low enough resistance - 0.15 Ohms (but only 6V rated, which is fine for normal operation as +5V power line, but maybe not good enough for "shorting to +12V" protection). Maybe I could even try to fit a larger 1206 footprint instead of 0805 (which gives us even better PTC fuses), but I'm not sure.