andreika-git/hellen-one #123: Flex input

rusefillc, 2021-12-17:
We need flex input on Hellen board

rusefillc, 2021-12-17:
https://github.com/rusefi/rusefi/wiki/Flex-Fuel says "we expect pull up on ECU"
@mck1117 @BeerMoneyMotorsports please spell out resistor range and voltage on https://github.com/rusefi/rusefi/wiki/Flex-Fuel

rusefillc, 2021-12-17:
@BeerMoneyMotorsports does any specific revision of any specific Miata boards support Flex already?

We shall put an explicit note on the PDF if we do!

BeerMoneyMotorsports, 2021-12-18:
Miata models that support Flex input.  To note, this has only been confirmed on the 90-95.  We need users to confirm functionality on the other boards. 

Model-- Header Pin    --    MCU     -- Resistor pullup
90/95 -- 2L(In Digital) -- GPIOF_3   -- 2.7k to 5v
96/97 -- 4B(In Digital) -- GPIOE_12 -- ?? to 5v
99/00 -- 4B(In Digital) -- GPIOE_12 -- ?? to 5v

96-00 uses pullup through the digital module.  i cannot identify what resistors are in use. 
[image]

rusefillc, 2021-12-18:
@andreika-git let's put a note at least on 90-95 schematics?