andreika-git/hellen-one #124: CAM mode for module-vr-discrete

rusefillc, 2021-12-21:
@mck1117 says there is an untested selectable pullup (ie, we can turn it off via another gpio)

@mck1117 can you provide a relevant snip?

@mck1117 what's the way to proceed in your opinion?

rusefillc, 2021-12-21:
Matthew Kennedy  11:20 PM
so cam vs vr

11:20
I think we probably could (edited) 
11:20
the VR interface doesn't care
11:20
the problem would come with the pullup resistor used for digital input

andrey  11:46 PM
on 2chan we can have a dip switch
11:46
on other stuff we can solder a resistor

Matthew Kennedy  11:46 PM
I don't like solder

andrey  11:46 PM
define "problem" i guess

Matthew Kennedy  11:46 PM
I designed for atlas an unproven selectable pullup (ie, we can turn it off via another gpio)

andrey  11:47 PM
how do we want to proceed?
11:47
breadboard test?

Matthew Kennedy  11:47 PM
it SHOULD work
11:47
I did ltspice test
11:47
which I probably trust

rusefillc, 2021-12-21:
https://github.com/rusefi/alphax-2chan/issues/9

mck1117, 2021-12-21:
[image]

mck1117, 2021-12-21:
When PULL_EN is high, IN1 is pulled up to V_PULLUP via the 2.7k resistor.  When PULL_EN is low, Q1002 goes open circuit and disconnects the pullup.

We would need a separate digital input circuit (series R, clamp diodes, etc) attached to the VR+/VR- pins (two digital inputs possible per VR!), with this circuit replacing the plain pullup resistor.

andreika-git, 2021-12-21:
That's exactly the same method we discussed yesterday with Andrey, but we talked about selectable 5v/12v pullups for tachometer!

Are you sure that the separate digital input would be required, and there is no way to use the vr-discrete as a "pass-through"  for hall sensors (with an external disconnectable pull-up)? I mean, we already used MAX992x as a pass-through for halls, right?

mck1117, 2021-12-21:
Max992x as a pass through had some quirks, particularly with the capacitor across the inputs having the potential to cause a bounce in the output on an edge. This has the same risk. It's better to use the proven digital input circuitry instead of trying to pass through the analog VR front end. It's so cheap and so little area.

andreika-git, 2021-12-22:
> We would need a separate digital input circuit (series R, clamp diodes, etc) attached to the VR+/VR- pins (two digital inputs possible per VR!), with this circuit replacing the plain pullup resistor.

We need to select the series resistance so that it wouldn't clamp the vr signal below zero. If the resistance isn't big enough, then the signal may become unsymmetrical... @mck1117 Do you think 20k is enough?

[image]

mck1117, 2021-12-22:
VR module already has its own clamps.

[image]

Since it's a "true" differential input, R758 pictured will pull both inputs (weakly) toward VCC/2, preventing asymmetric clipping.

mck1117, 2021-12-22:
tl;dr a single 10k will be just fine.  We could even use VR+ and VR- *both* as digital inputs.

mck1117, 2021-12-22:
Also, just to confirm, those buffers should be a 74*HC*2G17, not 74*LVC*2G17, since only the HC is specified for both input diodes.

[image]

andreika-git, 2021-12-22:
> tl;dr a single 10k will be just fine. 
The IN_CRANK+ is unclamped yet, and can potentially have big voltage spikes.. Do you think that one 1206 resistor voltage rating (200V) is enough? Maybe it's still more safe to put them two in series (4.7k+4.7k)

> We could even use VR+ and VR- _both_ as digital inputs.
Yes, but not on my current project - I'm afraid the board is too small to fit another disconnectable pullup circuitry for the second input...

> Proteus uses the 'HC2G17, and has been tested
Yep, Hellen uses it too.

mck1117, 2021-12-22:
> The IN_CRANK+ is unclamped yet, and can potentially have big voltage spikes

Huh? It's clamped inside the VR module already. A single 1206 is plenty. I've never seen more than tens of volts on a VR.

andreika-git, 2021-12-22:
Did you see the reference schematics for MAX992x?
[image]
Did you see the schematics of the Frankenso?
[image]
Did you see the schematics of the vr module?
[image]

The signal is clamped inside the vr-discrete module, but that's after the R752&R753 resistors, not before.
Let me also quote the Infineon paper (eg. for TLE4922): "_The output voltage of a VR-sensor can range from only a few Millivolts going up **to several hundreds of Volts** depending on air gap and tooth-frequency in front of the sensor._ "