andreika-git/hellen-one #126: TJR1443 schematics

rusefillc, 2021-12-31:
TJR1443
https://octopart.com/search?q=TJR1443&currency=USD&specs=0
is the best of available transceivers with wake-up feature, we shall try it on some of our designs

rusefillc, 2021-12-31:
Do we have space on alpha-x? Maybe on back side since that's a one-time test?

andreika-git, 2022-01-01:
Maybe we can try TJR1442AT first? It's the same [TJR114x family](https://www.nxp.com/products/interfaces/can-transceivers/can-with-flexible-data-rate/automotive-can-fd-transceiver-family:TJA144x), but in a SO8 package - pin compatible with our existing CAN module.

rusefillc, 2022-01-01:
https://octopart.com/search?q=TJR1442AT&currency=USD&specs=0 shows no availability comparing with https://octopart.com/search?q=TJR1443&currency=USD&specs=0

rusefillc, 2022-01-08:
@andreika-git can we test TJR1443 on alphax-4chan?