andreika-git/hellen-one #129: alphax-2 chan ibom issue

rusefillc, 2022-01-14:
https://rusefi.com/docs/ibom/alphax_2ch-a-ibom.html is not well

rusefillc, 2022-01-14:
Side note https://gitlab.com/kicad/code/kicad/-/issues/10427

andreika-git, 2022-01-15:
Some improvements were made: https://github.com/rusefi/alphax-2chan/commit/abc5a8ebcc6c41277835404d8beb879a369020f5
[image]
But still the "pcb" mode needs a solution for wrong VRML export...

andreika-git, 2022-01-15:
These commits contain the fix:
https://github.com/rusefi/alphax-2chan/commit/92fd5b5673dd5f4d12652d63d3fbda9657d96fd1
https://github.com/rusefi/alphax-2chan/commit/91e5ef016314e5647f6cc83b78ca2cee16ba263e
https://github.com/rusefi/alphax-2chan/commit/e9ce4495ee0449a3d88da4ec8a895c0278310a68
https://github.com/andreika-git/hellen-one/commit/f4a77038f289d80cee1175e8621dbb18583f7e1a
https://github.com/andreika-git/hellen-one/commit/87ed08e89f21c9c0805b78007eb04ac6608884f4
https://github.com/andreika-git/hellen-one/commit/15edebca0159f1d3145b61e1e0d41f87b1c1dd0e
https://github.com/andreika-git/hellen-one/commit/fcb62a786cf2ece60d0c5d966575e693b0ccbd8e
https://github.com/andreika-git/hellen-one/commit/8a70b219a5906cba4a520e92416ce6ff446d2ccc
https://github.com/andreika-git/hellen-one/commit/8cc300512e82af056219ab5d529753e834c27296

rusefillc, 2022-01-29:
progress was made

closed 2022-01-29 by rusefillc