andreika-git/hellen-one #139: Simplify MCU power supply architecture

mck1117, 2022-02-17:
The MCU module should have the following external power supply pins:
- 5V MCU supply input - powers MCU analog and digital
- V5A "aux" supply input - powers anything not-MCU
- V33 - 3.3v output from MCU module's regulator, supplied via 5V

Internal to MCU module:
- VCC -> 3.3v regulator generates MCU supply voltage
- VCC -> 3.3v reference chip (REF3033 or similar) generates voltage for VREF+. These references are not only more accurate than the regulators we use now, but are also far more stable over temperature changes, since we could realistically see -20 to +70 C
- 3.3v -> 3.3va ferrite bead powers MCU analog rail (not separate regulator)
- V5A powers opamps
- Retain existing VBUS -> VCC architecture

Rough list of changes:
- Remove U111, move L101 ferrite to between 3v rails instead of 5v.
- Do NOT internally power V5A rail, this allows us to turn off opamps externally.
- On "normal" boards, connect V5A to 5V externally.
- On "low power aware" boards, connect V5A to a separate supply that we can turn off at our leisure.

Why?
- More precise 3.3v reference improves analog accuracy and stability
- Using the same regulator for 3v and 3v_analog ensures meeting STM32's power sequencing requirements (today, if 5va is powered externally from a separate regulator, you _will_ violate the power sequencing and risk undefined behavior or damage)
- Externally split rails means it's simpler to do "low power" boards that need to turn off non essential logic

mck1117, 2022-03-17:
@andreika-git the change to use a polyfuse on the mcu module is not good - it means its now easier to exceed the maximum ratings for the STM32.

It doesn't allow a difference in voltage between VDD and VDDA, which is now possible if an external fault causes the polyfuse to trip.

[image]

Not to mention that losing the VDDA rail will cause the main PLL to unlock, halting the stm32.

mck1117, 2022-03-17:
[image]

rusefillc, 2022-10-17:
@Qwerty-OFF just a FYI

rusefillc, 2022-10-17:
We have REF3033 on WBO shall we reuse? https://github.com/andreika-git/hellen-one/issues/189

mck1117, 2022-10-17:
@rusefillc we should do it the other way around, put a ref3x33 in the MCU module, then have an INPUT on the wbo module to use the same reference as MCU

rusefillc, 2022-10-17:
Yes absolutely. But in boring real life https://github.com/andreika-git/hellen-one/issues/189 has a chance to happen much sooner.

rusefillc, 2022-12-05:
@Qwerty-OFF it's probably just a bit late to ask about this one?

rusefillc, 2022-12-07:
@Qwerty-OFF says that we have at least three separate reasons to change 

* 3.3 precision
* shutdown of op-amps
* fewer components

Since mm-0.1 seems ready my proposal is to implement this ticket within mm-0.2 later this month

mck1117, 2022-12-07:
Yes - those are all good reasons. The opamps right now I think are most why the the "sleep" current is so high.

dron0gus, 2022-12-07:
Few thoughts:
I use 2.5V as ADC reference voltage. Just because 5 / 2.5 = 2, while 5 / 3.3 = 1.51(51). I use same 0.5% resistors in top and bottom half of divider between OpAmps and MCU.
I use -0.6V and 5.5V to supply OpAmp. Not to care how good and linear OpAmps when close to rails. LM7705 for -0.6V and TLE7368E's 5.5V pre-regulator. If you going to have separate controllable supply for OpAmps 5.5V is reasonable.

For analog input diagnostic we should reserve some top ADC values for "short to battery" detection. On Subaru's stock ICU analog inputs have both pull-up and pull-down. Pull-up is something standard 1K and pull-down is hundreds of KOhm. So when sensor is disconnected ADC will see something < 5.0V, bit still out of valid range. And while it is short to battery ADC will see max possible value. This will need some additional support from SW.

mck1117, 2022-12-08:
> Not to care how good and linear OpAmps when close to rails.

The MCP6004/TP6004 are very good down to below 10mv. No sensors I've ever seen go below ~0.2v, so short to ground detection is easy enough. I've never seen it not work on any existing rusEFI board that uses 0v negative amp supply with MCP600x/TP600x. We had a batch accidentally made with LMV324, and those certainly had problems with both rails.

> For analog input diagnostic we should reserve some top ADC values for "short to battery" detection.

We could, yes, but I'm not sure of the value. I think the "separate opamp supply" is actually the "everything but MCU supply", so it probably does have to be 5.0v, not 5.5v/6v.

Besides, do we think there is additional value in detecting "short to 5v" vs "short to 12v"? We can detect that there is _some_ problem either way.

dron0gus, 2022-12-09:
> Besides, do we think there is additional value in detecting "short to 5v" vs "short to 12v"? We can detect that there is _some_ problem either way.

AFAIR none of stock ECU differ short to +12 and short to +5. So everything close to max ADC value is short to battery.
I'll measure pull-up and pull-down on stock Subaru ECU on temperature and MAF inputs and get back.

Qwerty-OFF, 2022-12-27:
@mck1117 please check
[image]

mck1117, 2022-12-29:
That schematic looks OK to me. Is there a pulldown on PWR_EN? Reminder that the STM32 is effectively under reset while asleep, so the pins could be floating, and static will build up and turn on the aux rails.

rusefillc, 2023-02-11:
Status update: I have https://github.com/rusefi/rusefi/wiki/Hellen-125-Honda-K board based on https://github.com/andreika-git/hellen-one/tree/master/modules/mega-mcu144/0.2 in my hands!

Current state https://github.com/rusefi/hellen125honda-issues/issues/19 and now seems like great time to ask if any software was expected :)

Let's document key points at https://github.com/andreika-git/hellen-one/wiki/module-mega-mcu https://github.com/andreika-git/hellen-one/wiki/module-power12-and-5

like what is PG_5VP? etc