andreika-git/hellen-one #143: why is FATAL led green on mcu144 boards?!

rusefillc, 2022-03-12:


rusefillc, 2022-03-12:
PDF says 
[image]

rusefillc, 2022-03-12:
https://rusefi.com/docs/ibom/alphax_2ch-b-ibom.html

ibom says D104 is D104 and D103 is red

[image]

rusefillc, 2022-03-12:
while we are fixing this I suggest usability improvement:

instead of LED1-4 let's have LEDR LEDG LEDB LEDY

rusefillc, 2022-03-12:
@andreika-git what's your take on this issue? does it look like schematics is wrong while ibom is right?

andreika-git, 2022-03-13:
[image]

Both the schematics and iBOM are correct. The label positions on schematics are confusing: a designator is on top of a component and value is on bottom.

rusefillc, 2022-03-13:
green LED was a SW issue https://github.com/rusefi/rusefi/commit/9de0c10f24f9f587f0179a4230439f8a1fd08144