andreika-git/hellen-one #145: Combined footprint for both kinds of TLS115

rusefillc, 2022-03-23:
is it possible to combine TSON-10 TLS115D0LDXUMA1 and SOIC-8 TLS115B0EJXUMA1?

that would be beneficial in current chip shortage times - I can hand solder my SOIC-8 stock while waiting for JLC to get  TSON-10 parts

andreika-git, 2022-03-24:
P.S. https://github.com/andreika-git/hellen-one/blob/master/kicad/footprints/PG-TSON-10-2.kicad_mod

rusefillc, 2022-03-24:
But we can probably use TSON-sized thermal-pad for both? @mck1117 opinions?

I agree that we can manually manage between batches but shared footprint is so much easier on humans!

andreika-git, 2022-03-24:
The SOIC package itself (which is actually "PG-DSO-8 EP" not SOIC) has the big-sized thermal pad on its bottom with would overlay and short the pads of TSON.

mck1117, 2022-03-24:
I would not try to do shared footprint. The pins of the smaller footprint nearly (or do) overlap the exposed pad on the larger footprint - so you'd short out every pin by installing the larger chip.

We can always just put two footprints next to each other...

closed 2022-12-27 by Qwerty-OFF