andreika-git/hellen-one #148: MCU io5 readability

rusefillc, 2022-03-30:
[image]

[image]

[image]

rusefillc, 2022-03-30:
we have out_io1 and io1 which are different things, io1 which is connected to io5, io5 and out_io5 which are different things

that's confusing

let's rename both IO1 and is IO5 which are same thing to GP1 maybe? any other ideas?

andreika-git, 2022-03-31:
Yes, we can rename IO[1|5]+ to GP1+. Do we need a new mcu revision for this?

rusefillc, 2022-03-31:
Nope i think same mcu revision perfectly fine as long as zero copper changes

rusefillc, 2024-07-28:
@Qwerty-OFF please rename the not-"OUT" IO to GP_IOx

[image]