andreika-git/hellen-one #161: Actually use Docker

rusefillc, 2022-05-18:
Current status: we have self-contained docker sets up all dependencies and runs a functional test see happy https://github.com/andreika-git/hellen-one/blob/master/.github/workflows/test-check-all.yaml

Next step: we need some sort of file integration/mount external file system/volume in order to invoke https://github.com/rusefi/alphax-2chan/blob/main/step2_copy.sh and https://github.com/rusefi/alphax-2chan/blob/main/step3_create_board.sh within docker maybe using https://github.com/andreika-git/hellen-one/blob/master/step2-run-docker.sh

ZHoob2004, 2022-05-18:
https://github.com/ZHoob2004/alphax-2chan/commit/7b87e7bb00ba92284ece0bde7cd8459fdaad8c2e

example implementation to use existing container as a build tool

In short, it mounts alphax-2chan at /alphax-2chan in the container then uses the --entrypoint parameter to run either bash or python depending on the task required

[step1_build_hellen-one_docker.sh](https://github.com/ZHoob2004/alphax-2chan/blob/main/step1_build_hellen-one_docker.sh) could be removed if we push hellen-one to docker hub. (like using GHA)

In that case, later commands would need minor revision to use (for example) andreika/hellen-one instead of the locally tagged builds

rusefillc, 2022-05-18:
thank you @ZHoob2004 !!!

rusefillc, 2022-05-21:
Wow, just wow.

closed 2022-05-21 by rusefillc