andreika-git/hellen-one #167: TLS115 is important but chip shortage is forcing us to have both packages supported

rusefillc, 2022-06-10:
Right now TLS115D0LDXUMA1 in TSON-10 package is more available than TLS115D0EJXUMA1 in PG-DSO-8 package

flexibility of ordering is desired, please investigate if a hellen module for just one part with two variations is a way to achieve

rusefillc, 2022-06-11:
[image]

rusefillc, 2022-07-25:
See https://github.com/andreika-git/hellen-one/commit/0fe5bfe417e1f6757a138ce7c60db6277b0a5df2

rusefillc, 2022-08-23:
Looks like we are just handling this manually for now and hellen module is not needed?

@Qwerty-OFF any opinions?

Qwerty-OFF, 2022-08-23:
Not much work here, perhaps we should make a double-sided hellen module in which two footprints will be compactly combined.

rusefillc, 2022-08-23:
I like this idea!

hand-assembled SOIC should be on the back and TSON-10 should be on top side