andreika-git/hellen-one #171: Expose all stm32 pins on mcu-144

rusefillc, 2022-07-26:
How come we have a few unconnected pins

[image]

[image]
[image]
[image]

[image]

rusefillc, 2022-07-26:
@andreika-git can you please remind? is this related to 2 layer limitation of the module or else?

andreika-git, 2022-07-30:
> @andreika-git can you please remind? is this related to 2 layer limitation of the module or else?

Related to very dense component layout and lots of vias (which are pass-through all layers). 
4 layers version *could* possibly help to expose some of the hidden pins, but no guarantees...

rusefillc, 2022-07-30:
https://github.com/rusefi/alphax-4chan/issues/51

rusefillc, 2022-10-05:
[8chan ](https://github.com/rusefi/alphax-8chan/issues/22) is asking for this change on MCU 0.7

andreika-git, 2022-10-20:
[image]

[image]

The new exposed pins are:
- IO5 = PG15
- IO6 = PD8
- IO7 = PF15
- IO8 = PE10
- IO9 = PF2 (not recommended! reserved for future Board-ID expansion)

Currently all 144 pins are used!

The mcu-0.7 module (176-pin) was also modified to support the new symbol/footprint (and the same pins).

closed 2022-10-20 by andreika-git

rusefillc, 2022-10-20:
or even remove it from footprint?