andreika-git/hellen-one #176: double-sided version of WBO maybe with integrated CAN

rusefillc, 2022-08-06:
problem statement: fit more stuff by using double-sided assembly

WBO always goes with CAN

can we reduce combined space by going double-sided?

Qwerty-OFF, 2022-08-06:
Without changes in WBO module it is impossible, bottom side occupied by tracks.

rusefillc, 2022-08-06:
Right, i mean to change the module

Qwerty-OFF, 2022-08-07:
Is only CAN move to bottom or other WBO components also for minimization?

rusefillc, 2022-08-07:
any component where it's reasonable for electrical reasons could go on the back side in order to reduce total rectangle size

@mck1117 FYI

mck1117, 2022-08-08:
do we want to change random things, or seek some stability in the world?

rusefillc, 2022-08-08:
@mck1117 please reduce noise.

if are looking for a clarification on reasons beyond what should already be available to you and obvious please just ask explicitly.

rusefillc, 2022-10-12:
@Nadezhda-n можешь попробовать?

rusefillc, 2022-10-17:
See also https://github.com/andreika-git/hellen-one/issues/189

rusefillc, 2022-10-24:
@Nadezhda-n получается первый шаг - это к kicad схеме WBO добавить AD схему CAN?

Nadezhda-n, 2022-10-25:
[image]
We need to add CANL, CANH, CANVIO on the module

Nadezhda-n, 2022-10-27:
ok
[image]
[image]

closed 2022-12-27 by Qwerty-OFF