andreika-git/hellen-one #180: placement file generator should fail instead of producing invalid file

rusefillc, 2022-09-01:
https://github.com/rusefi/hellen-gm-e67/ rev B board is invalid because of https://github.com/rusefi/hellen-gm-e67/issues/81

Current behavior: hellen framework has successfully generated a partial placement file https://github.com/rusefi/hellen-gm-e67/blob/main/boards/hellen-gm-e67-b.fab/board/hellen-gm-e67-b-CPL.csv - all the components below R54 in BOM are missing in placement file :(

Expected behavior: hellen framework should fail to generate board when PCB has components not present in schematics

rusefillc, 2022-09-02:
Let's test on e67

so test of this #180 depends on #181

andreika-git, 2022-09-10:
Now the board files won't be created if any of the designators are absent.
Also added an error message:

"Error! Designator XXX was not found in the BOM file!
Error: Process completed with exit code 2."

closed 2022-09-10 by andreika-git

rusefillc, 2022-10-23:
Not sure if related but still some issues on E67 https://github.com/andreika-git/hellen-one/issues/191