andreika-git/hellen-one #185: Супермодуль

Qwerty-OFF, 2022-10-11:
Что сложить в супермодуль:
- mcu
- input  (и светодиоды индицирующие состояния входов?)
- ignition (и его светодиоды?)
- can
- sd наверно тоже всегда должен быть?
- засунуть туда же барометр и акселерометр?
- кнопки boot и reset чтобы больше не помнить о том, что их нужно ставить
- может быть еще и dinpu4, если вдруг покажется что у нас есть много места

Посмотрел на мультиплексирование аналоговых входов в 8chan, это уже обдуманное решение что нужно три мультиплексора 2:1? В природе существуют другие варианты, в том числе когда сразу несколько мультиплексоров в одной микросхеме чуть-чуть экономят место.

По поводу формы, размещения падов модуля и всего такого:
[image]
1, 2. - пады от модулей input и ign должны быть со стороны основного разъема
3. CAN и возможно SD въезжают внутрь супермодуля, почти все hellen платы скомпонованы как на этой картинке, соответственно высоту супермодуля можно увеличивать вплоть до суммы высот input + mcu + sd. За счет двухстороннего монтажа должно получиться сэкономить высоту. В 2chan и 4chan по высоте сейчас влезет только супермодуль размером input + mcu.
4. Пады которые предназначаются для модуля output можно сдвинуть немного ниже, чтобы соединяющие линии шли горизонтально, а не по диагонали как на картинке.
5. Все прочие пады вынести в сторону правого нижнего угла (stlink, "лишние" ноги для прототипирования, шины spi и i2c), либо верхнего правого угла (горстку ног для подключения ETB, USB, может еще что-то). Нижнюю сторону хочется освободить от падов вообще, чтобы смело размещать под ней всякие вещи типа зоны прототипирования, WBO и т.п.
В общем и целом расстановка на платах hellen выглядит так, что у нас мало места по вертикали и чуть больше по горизонтали, т.е. приоритетно уменьшать сначала высоту, а потом ширину.
Само собой это просто хотелки, а не требования, надо смотреть насколько грустно будет делать трассировку.

Другие идеи не касающиеся непосредственно супермодуля:
- объединить Power12, Power5 и TLS115 в одном модуле, кажется по одиночке они нигде не встречаются;
- двухсторонний output, в котором пады будут расположены чуть более удобно (сейчас часть падов предназначенных для основного разъема смотрит в сторону mcu)
- подумать о желании сделать двухсторонние WBO+CAN, VR-discrete, knock, что-нибудь еще

rusefillc, 2022-10-12:
* Внешний st-link в реальности не использовался ни разу вообще, TC2030 в кризисной ситуации хватит

* двухсторонние WBO+CAN
* VR-discrete
* knock

точно хочется и видимо стоит начинать просить Надю попробовать

Qwerty-OFF, 2022-10-12:
Кстати, можно компоненты размера 0603 менять на 0402 в тех местах где их точно никто и никогда не трогает паяльником вручную.

rusefillc, 2022-11-11:
Новая версия списка

* mcu
* aux mux 
* input без светодиодов
* can
* sd 
* кнопка boot точно
* dinpu4, в котороый нужно добавить https://github.com/andreika-git/hellen-one/issues/193 и тестовые площадки
* тестовые площадки?

* кнопка reset под вопростом
* барометр и акселерометр точно нет
* ignition всё-таки нет

rusefillc, 2022-11-14:
@Qwerty-OFF точно без VR и точно без защиты от высокого напряжение

rusefillc, 2022-11-26:
Secret preview

[image]

rusefillc, 2023-02-09:
@qwerty-off time to close?

closed 2023-05-05 by Qwerty-OFF