andreika-git/hellen-one #200: Megamodule: SENT considerations around MCU

rusefillc, 2022-11-20:
https://en.wikipedia.org/wiki/SENT_(protocol) is the future, while it's pretty far Hellen should definitely support at least one input maybe two

on the software side at the moment we use STM32 ICU (input capture unit) of TIMER, ChibiOS limitation is that we can only use channels 1 and 2

At the moment the only proven hardware combination is MRE with C54 removed see https://github.com/rusefi/rusefi/tree/master/firmware/hw_layer/drivers/sent

rusefillc, 2022-11-20:
@dron0gus FYI

https://github.com/rusefi/rusefi/pull/4270

rusefillc, 2022-11-20:
following pins where channel is 0 or 1 are best candidates. Not sure why A5 is not on the list but this list is probably mostly correct

```
static expected<stm32_pwm_config> getConfigForPin(brain_pin_e pin) {
	switch (pin) {
#if STM32_PWM_USE_TIM1
	case Gpio::A8:  return stm32_pwm_config{&PWMD1, /*channel*/0, /*AF*/1};
	case Gpio::A9:  return stm32_pwm_config{&PWMD1, /*channel*/1, /*AF*/1};
	case Gpio::A10: return stm32_pwm_config{&PWMD1, 2, 1};
	case Gpio::A11: return stm32_pwm_config{&PWMD1, 3, 1};

	case Gpio::E9:  return stm32_pwm_config{&PWMD1, 0, 1};
	case Gpio::E11: return stm32_pwm_config{&PWMD1, 1, 1};
	case Gpio::E13: return stm32_pwm_config{&PWMD1, 2, 1};
	case Gpio::E14: return stm32_pwm_config{&PWMD1, 3, 1};
#endif
#if STM32_PWM_USE_TIM2
	case Gpio::A15: return stm32_pwm_config{&PWMD2, 0, 1};
	case Gpio::B3:  return stm32_pwm_config{&PWMD2, 1, 1};
	case Gpio::B10: return stm32_pwm_config{&PWMD2, 2, 1};
	case Gpio::B11: return stm32_pwm_config{&PWMD2, 3, 1};
#endif
#if STM32_PWM_USE_TIM3
	case Gpio::B4:  return stm32_pwm_config{&PWMD3, 0, 2};
	case Gpio::B5:  return stm32_pwm_config{&PWMD3, 1, 2};

	case Gpio::C6:  return stm32_pwm_config{&PWMD3, 0, 2};
	case Gpio::C7:  return stm32_pwm_config{&PWMD3, 1, 2};
#endif
#if STM32_PWM_USE_TIM4
	case Gpio::B6:  return stm32_pwm_config{&PWMD4, 0, 2};
	case Gpio::B7:  return stm32_pwm_config{&PWMD4, 1, 2};
	case Gpio::B8:  return stm32_pwm_config{&PWMD4, 2, 2};
	case Gpio::B9:  return stm32_pwm_config{&PWMD4, 3, 2};

	case Gpio::D12: return stm32_pwm_config{&PWMD4, 0, 2};
	case Gpio::D13: return stm32_pwm_config{&PWMD4, 1, 2};
	case Gpio::D14: return stm32_pwm_config{&PWMD4, 2, 2};
	case Gpio::D15: return stm32_pwm_config{&PWMD4, 3, 2};
#endif
#if STM32_PWM_USE_TIM5
	case Gpio::A0:  return stm32_pwm_config{&PWMD5, 0, 2};
	case Gpio::A1:  return stm32_pwm_config{&PWMD5, 1, 2};
	case Gpio::A2:  return stm32_pwm_config{&PWMD5, 2, 2};
	case Gpio::A3:  return stm32_pwm_config{&PWMD5, 3, 2};
#endif
#if STM32_PWM_USE_TIM8

#if !STM32_PWM_USE_TIM3
	// If TIM3 is not used, put these pins on TIM8 instead..
	// See https://github.com/rusefi/rusefi/issues/639
	// See https://github.com/rusefi/rusefi/pull/3032
	case Gpio::C6:  return stm32_pwm_config{&PWMD8, 0, 3};
	case Gpio::C7:  return stm32_pwm_config{&PWMD8, 1, 3};
#endif

	case Gpio::C8:  return stm32_pwm_config{&PWMD8, 2, 3};
	case Gpio::C9:  return stm32_pwm_config{&PWMD8, 3, 3};
#endif
#if STM32_PWM_USE_TIM9
	case Gpio::E5:  return stm32_pwm_config{&PWMD9, 0, 3};
	case Gpio::E6:  return stm32_pwm_config{&PWMD9, 1, 3};
#endif
	default: return unexpected;
	}
};
```

Qwerty-OFF, 2022-11-20:
Supported inputs:
AUX2
AUX4
O2S
O2S2
Supported outputs:
OUT_IO2
OUT_IO12
OUT_PWM1
OUT_PWM2
OUT_PWM3
OUT_PWM8
Other supported IO:
SPI3_CS

rusefillc, 2022-11-20:
Let's only mark on the symbol to see which ones are ICU, looks like we have options for later