andreika-git/hellen-one #207: mcz33810 module

rusefillc, 2022-12-03:
Definitely no IGBT on-board, maybe provisions for external?

rusefillc, 2022-12-03:
@dron0gus my KIT33810EKEVME order from 10 days ago is in a very unclear state :(

@dron0gus can you cut relevant portion of your PDF to give @Nadezhda-n a head start?

See also https://github.com/rusefi/rusefi-hardware/tree/main/quad-igbt

dron0gus, 2022-12-04:
[image]

rusefillc, 2022-12-05:
@dron0gus Question on C5 see https://github.com/rusefi/rusefi-hardware/issues/159#issuecomment-1336579958

rusefillc, 2023-02-02:
Depends on https://github.com/rusefi/kicad6-libraries/issues/9

rusefillc, 2023-02-09:
Oh this one is used by https://github.com/mck1117/atlas

[image]

rusefillc, 2023-11-16:
depends on https://github.com/rusefi/rusefi-hardware/issues/255