andreika-git/hellen-one #243: Root cause of VR-discrete 0.2 Gerber edge maybe documentation issue?

rusefillc, 2022-12-15:
See https://github.com/andreika-git/hellen-one/issues/238#issuecomment-1353523007

Nadezhda-n, 2022-12-15:
Полагаю, что для модулей это не нужно, но, на самом деле, не понимаю, зачем это нужно для любых других плат :)
А так, да "Root cause of VR-discrete 0.2 Gerber edge maybe documentation issue?" - прямоугольник на всех герберах и есть край модуля, на пред версии, как и в других платах, его нет :)

Qwerty-OFF, 2022-12-15:
Кикад когда-то переименовал эту галочку с "не рисовать край платы на всех слоях" на "рисовать край платы на всех слоях". Видимо из-за этого возникла путаница. В руководстве по сути все правильно, но не соответствует свежему кикаду.
[image]

Nadezhda-n, 2022-12-16:
> Кикад когда-то переименовал эту галочку с "не рисовать край платы на всех слоях" на "рисовать край платы на всех слоях". Видимо из-за этого возникла путаница. В руководстве по сути все правильно, но не соответствует свежему кикаду. [image]

Ха-ха
Спасибо :)

closed 2022-12-17 by Qwerty-OFF