andreika-git/hellen-one #245: VR discrete issues

mck1117, 2023-01-03:


mck1117, 2023-01-03:
The bodge I did on the bench:
[image]

ElDominio, 2023-01-05:
Would I not have to put on at pin3 too? Since I see we're putting a cap across 2IN+ and GND, wouldn't one be needed between 1IN+ and GND too? just for my info

rusefillc, 2023-01-05:
@Qwerty-OFF can you help with ^^^?

ElDominio, 2023-01-05:
Yaris with 10nf cap across 4 and 5 still has issues

ElDominio, 2023-01-05:
[changingtocamactivewithinvertyaaris.zip](https://github.com/andreika-git/hellen-one/files/10355843/changingtocamactivewithinvertyaaris.zip)

I changed from not using cam to using cam

rusefillc, 2023-01-05:
@ElDominio please clarify

"VR discrete issue" not resolved 
or 
"VR discrete issue" impossible to test because of CAM issue 
or 
"VR discrete issue" resolved, now separate issue?

mck1117, 2023-01-05:
he forgot to add a second cap to the cam channel, and is going to test on a rev E so he can get an SD log

ElDominio, 2023-01-06:
Yaris
RevE with 10nF capacitors on TLV3202

[OsciLogandNormalLogwithcapsRevE.zip](https://github.com/andreika-git/hellen-one/files/10356445/OsciLogandNormalLogwithcapsRevE.zip)
[re_15.zip](https://github.com/andreika-git/hellen-one/files/10356446/re_15.zip)

ElDominio, 2023-01-06:
[image]

mck1117, 2023-01-06:
Cap seems to have made an improvement to crank signal:

before on left, after on right

Left has very not-50% duty cycle, falling edge comes very soon after rising edge. Right has near 50/50 duty, which is expected for VR signal with many teeth (like a 36-2 crank)

[image]

rusefillc, 2023-01-06:
https://github.com/andreika-git/hellen-one/issues/247 for actual change

let's start by improving newer 2 sided version? we might end up making improved one sided version as well I guess

closed 2023-01-22 by rusefillc

ElDominio, 2023-01-22:
We might need another cap according to @mck1117 , @rusefillc . Pending testing

mck1117, 2023-01-22:
> We might need another cap according to @mck1117 , @rusefillc . Pending testing

the updated schematic has both caps, like the original little test board I did