andreika-git/hellen-one #247: vr-discrete 0.3: Add 10nF between GND and 2IN+

rusefillc, 2023-01-06:
Please take vr-discrete)/0.2 add one capacitor and make that vr-discrete)/0.3

See #245 for details, we need one capacitor added to schematics and gerbers

rusefillc, 2023-01-06:
@Nadezhda-n can you take care of this?

mck1117, 2023-01-06:
I am not yet 100% sure that’s the only change required to fix issues. We can go ahead and draft the change, but don’t commit to that yet.

mck1117, 2023-01-07:
Go ahead and add a cap between pin 2 and ground (in addition to pin 5). I've never seen it happen on the top half, but it's theoretically susceptible to the same oscillation.

mck1117, 2023-01-07:
WAIT a second, those caps were already present on the original design, why did they disappear on the hellen module!?!?!

Little test board I made:

[image]

Hellen module:

[image]

notice C9/C10 are exactly what we need to add, but were already there

rusefillc, 2023-01-07:
hellen1-vr-discrete is one of the few kicad-based modules https://github.com/andreika-git/hellen-one/tree/master/kicad/modules

rusefillc, 2023-01-22:
@mck1117 please review and confirm if we can fab a batch of boards

mck1117, 2023-01-22:
0.4 schematic looks OK

closed 2023-01-22 by rusefillc