andreika-git/hellen-one #251: Any chance to make WBO module less weird?

rusefillc, 2023-01-10:
4.9 heater wires are "heater+" and "12 raw" is not that a bit confusing?

[image]

Snippet of https://github.com/rusefi/rusefi_documentation/blob/master/Hardware/Hellen/hellen125honda-a-schematic.pdf

rusefillc, 2023-01-10:
@mck1117 @dron0gus any ideas for at least footprint comments to reduce confusion?

rusefillc, 2023-01-11:
Let's make a compatible wbo-0.4 change by adding kicad notes to the symbol:

"LSU_H-" add blue note "power GND"
"GND" add blue note "analog/digital"
"LSU_H+" add blue note "heater negative"

Qwerty-OFF, 2023-01-11:
[image]

closed 2023-01-18 by Qwerty-OFF