andreika-git/hellen-one #270: A dream: less custom InteractiveHtmlBom

rusefillc, 2023-03-15:
We've started with https://github.com/openscopeproject/InteractiveHtmlBom and done _something_

Open question what exactly have we done.
Real question is if we should (or maybe should not) PR something to upstream? Maybe just encapsulation of view vs engine or some sort of plugin mechanism or else?

rusefillc, 2023-03-15:
@chuckwagoncomputing

chuckwagoncomputing, 2023-03-16:
It looks like this repo uses the original openscopeproject/InteractiveHtmlBom as a submodule, though a 2.5 year old snapshot of it. Are there in fact changes happening somewhere?

rusefillc, 2023-03-16:
maybe this ticket here should be "explain how do we produce https://rusefi.com/docs/ibom/alphax_4ch-f-ibom.html from not kicad files"

Separate https://github.com/andreika-git/hellen-one/issues/273 created