andreika-git/hellen-one #294: kicad vs jlcpcb orientation

rusefillc, 2023-05-11:
let's revisit orientation drama

rusefillc, 2023-05-11:
https://forum.kicad.info/t/is-there-a-simple-way-to-update-all-my-footprints-to-use-the-jlcpcb-orientations/38931

rusefillc, 2023-05-12:
most recent example: on 4chan rev G question was if D418 was placed right or not

Qwerty-OFF, 2023-05-16:
https://github.com/matthewlai/JLCKicadTools/tree/master уже живет внутри Hellen-one https://github.com/andreika-git/hellen-one/tree/master/bin/jlc_kicad_tools

У нас есть БД с поворотами для футпринтов, проблема в том что разные компоненты с одним и тем же футпринтом требуют разный поворот. Т.е. увидим ту же историю что здесь https://github.com/matthewlai/JLCKicadTools/issues/28 

Нужно знать как компонент лежит в ленте, а значит такой словарик нужно вести не для футпринтов, а для LCSC-кодов компонетов

rusefillc, 2023-05-16:
внезапно! новая проблема https://github.com/andreika-git/hellen-one/issues/296

ну и пинганул товарищей в https://github.com/matthewlai/JLCKicadTools/issues/28

Qwerty-OFF, 2024-05-29:
Работает и выглядит очень полезным и нужным