andreika-git/hellen-one #295: Mega-mcu144 (0.1-0.4): wrong partnumber for 10k resistor

andreika-git, 2023-05-16:
C11616 instead of C25744

https://github.com/andreika-git/hellen-one/commit/4b77ad9d21666b3cd91e542e3dc999dc61a85707

Qwerty-OFF, 2023-05-16:
Полный список подверженных резисторов:
* R124, R126, R132, R135, R136, R137, R138, R139, R228, R229, R230, R231, R232, R233, R234, R235, R246, R247, R248, R249, R250, R251, R252, R253 - это подтяжки в input. На конкретных платах они могли быть изменены на другие номиналы с помощью replace-механизма. В случае сомнений в работе какого-либо входа ММ нужно смотреть PN соответствующей подтяжки в БОМ на конкретную плату.
* R102 - не дающий подпитывать цепь питания USB от 5V малым током 
* R103 - подтяжка SPI2, заменить на 10k при использовании SPI2
* R105 - подтяжка SPI2, заменить на 10k при использовании SPI2
* R108 - подтяжка в цепи управления переключаемыми 5В. Заменить если есть подозрение что напряжение не отключается или отключается долго
* R109 - подтяжка в цепи управления переключаемыми 3.3В. Заменить если есть подозрение что напряжение не отключается или отключается долго
* R121 - подтяжка nReset. Есть встроенная в контроллере, вероятно все безопасно
* R127 - подтяжка boot0. Если вдруг контроллер входит в режим программирования, заменить
* R128 - подтяжка boot1
* R141 - подтяжка линии PWR_EN. Менять если не выключаются все питания, в том числе TLS115 в спящем режиме (когда нога контроллера висит в воздухе)
* R142 - подтяжка линии управления мультиплексорами. Если при переходе из "1" в "High-Z" мультиплексор не переключается в каналы аналогичные "0"

Qwerty-OFF, 2023-10-02:
Исправлено в 0.5

closed 2023-10-02 by Qwerty-OFF